32blit/32blit-sdk #554: Power off / turning off flashing power light

mikerr, 2021-01-15:
Those of us without a battery see a constantly flashing power LED.

Selecting "power off" in the firmware menu used to turn the LED off.
Well it  still does - but now it also stops you going back into the menu ever again until a power disconnect !

Suggest a return to the old functionality, ora new menu item to just blank the LED ?

Gadgetoid, 2021-01-15:
Ooofff... this sounds like a bug. "Power off" effectively puts the power chip into "Shipping Mode" and shouldn't affect anything at all if you're connected to external power. Though that said, "Power Off" should *actually* power the device off.

I think it's possible to clear the fault on the battery charge IC, or at least tell it to stop wildly flashing the LED. We changed this for the final units, wiring the LED to the STM32 instead.

Daft-Freak, 2021-01-15:
Hmm, for me closing the menu causes the LED to come back on... which I guess is because that button is _also_ the one used to turn the device back on...

Gadgetoid, 2021-01-15:
Ooof, is *this* what you mean by not being able to get back into the menu @mikerr?

Will have to look into convincing the charge controller to... shhh.

mikerr, 2021-01-16:
No, here's a full workflow:

1.<power LED flashing>
2. Enter firmware menu 
3. Select "power of"f
4. <power LED off>
5. Select "exit game"

You're now back in launcher with LED off. (yippee!)

You can launch games, but any press of the home button is ignored, 
so you can't :
- open firmware menu.
- exit games with a long press

Still ignored after pressing reset, only pulling power returns back to normality (with flashing LED again !)

This change happened somewhere around beta 0.14 as far as I can tell.

Daft-Freak, 2021-01-20:
Yep, that definitely happens. Home button is entirely non-functional.

It's not a software bug though. With no battery and the charge IC off, there's no VBAT... which is what the home button uses...

Looks like this was "broken" by the button switch. (More of a weird edge-case).

(Interestingly, if I turn the brightness down by one step on my blit the LED stops flashing...)

mikerr, 2021-01-21:
Ha, turning backlight to 90% stops the LED flashing quite so rapidly, then it just blinks every second or so...

Daft-Freak, 2025-01-27:
I don't think there's anything that can be fixed here. (Not in software anyway)

closed 2025-01-27 by Daft-Freak