32blit/32blit-sdk #567: Add saving of menu setting states (QoL enhancement)

ThePythonator, 2021-01-25:
This is more of a QoL suggestion than an issue, but it would be useful if the firmware menu saved the settings you adjust (such as volume level, muted, backlight level etc).

I'm not sure if the firmware has access to the same sort of saving systems as most .blit files do, but if it does it shouldn't be hard to save these values in the same way as other games save their data.

Gadgetoid, 2021-01-25:
Afaik firmware does have access to save states the same way games do, and it should probably save state.

However- volume level/backlight level are *supposed* to be persistent. There's a little portion of RAM reserved for this that doesn't get initialized at startup but does require constant power. And... I think I've just answered my own line of reasoning as to why *that* doesn't work so well.

We need "power off" to actually be a low-power sleep mode for the persistent RAM saving to be any use.

But yes, saving to disk is the sensible definitely-don't-lose-these-settings choice.

Daft-Freak, 2021-01-29:
It would probably just be a case of `save_game(persist)` in a few places (before power off and game switches?), though that might end up pulling a bunch of file related code into the firmware... (and `load_game` in the bit that sets the initial values)

(I like the fact that "Power off" actually turns _everything_ off though... guess that's more about the "sleep mode" issue though)