32blit/32blit-sdk #568: Buffer audio instead of firing callback at 22,000Hz

Gadgetoid, 2021-01-25:


Daft-Freak, 2021-01-25:
Checking if the DAC should be enabled every sample probably isn't great either... https://github.com/pimoroni/32blit-beta/blob/master/32blit-stm32/Src/sound.cpp#L22

Gadgetoid, 2021-01-26:
I did some sleuthing for the old audio code that used DMA and a buffer, and recalled that I never did manage to get it working. The system would just lock up solid and I had to abandon all hope in favour of... this travesty. So... yep. Long overdue a fix.

zenodante, 2021-01-27:
I will work on it, after I receive the hardware. Current my h750-dk boards don't have a DAC audio setup.  \(Q_Q)/

Daft-Freak, 2025-01-17:
I ended up figuring this out (eventually): https://github.com/Daft-Freak/32blit-beta/commit/9b8d4a268d107b3d35ba43957482833acba02414 (Very lightly tested)

Mostly because I suspected that this was why audio was weird in my new DOOM port...