32blit/32blit-sdk #574: Volume is too LOUD

ali1234, 2021-02-01:
In my opinion, the volume slider doesn't have enough range at low volume levels, and too much at high levels. Basically, the lowest volume you can set is too loud for a quiet room.

Volume handling code is here: https://github.com/pimoroni/32blit-beta/blob/master/32blit-stm32/Src/32blit.cpp#L215

This code attempts to do a logarithmic volume control. However, it seems to be actually doing anti-log. I made a graph to plot the output from this function:

https://www.desmos.com/calculator/rzxgdduvi7

H is the value of volume_log_base. In the existing code it equals 2. This makes the volume increase quickly at first, and then slowly at higher volumes. This seems backwards to me: it should increase slowly at first, and then quickly at higher volumes. In order to achieve that, H must be less than 1. (At 1 it would behave linearly, except that it causes a division by zero.) From experimentation I have found that a value around 0.04 is needed to get the lowest volume setting suitable for a quiet room.

tinwhisker, 2021-02-02:
Would an 's' curve formula be more suited? This would give greater granularity at either end.

ali1234, 2021-02-15:
Not really. Currently there is no discernable difference between the maximum and max-1 volume, while min+1 is about twice as loud as minimum volume. The backlight control behaves in a very similar way too.