32blit/32blit-sdk #584: Serial

ajh123, 2021-02-05:
Is there a way to receive data from the serial debugger

ajh123, 2021-02-05:
because we can send data ```blit::debug("Hello, world!");``

Daft-Freak, 2021-02-05:
The other direction is mostly used to send commands to the firmware, but if you prefix with `32BLUSER` + uint16 length, you can receive data using the "multiplayer" `blit::message_received` callback.

ajh123, 2021-02-05:
Whats the baud rate to connect via Python would 9600 be fine?

ajh123, 2021-02-05:
i tried. this python
```
import serial
from time import sleep
ser = serial.Serial('/dev/tty.usbmodem3271357530391', 2000000, timeout=2, xonxoff=False, rtscts=False, dsrdtr=False) 
ser.flushInput()
ser.flushOutput()
while True:
  data_raw = ser.readline()
  print(data_raw)
  msg = 'Hello?'
  ser.write(bytes('32BLUSER'+str(len(msg))+msg, 'ascii'))
```
but the 32blit dose not seem to receive 'Hello?', however the python can receive it's messages

Daft-Freak, 2021-02-05:
The length is a uint16, so you want something like `ser.write(b'32BLUSER'+ struct.pack('<H', len(msg)) + msg.encode('ascii'))`. 

Baud rate doesn't matter (but I guess you worked that out already :smile:)

ajh123, 2021-02-05:
where do i import ```struct```. from im in Python3

ajh123, 2021-02-05:
found it, it works now, thanks!

Gadgetoid, 2021-03-25:
I wonder if this - receiving debug output and sending arbitrary data back - should be a feature of the tools. You could then just import the relevant part of the tools in your code and do something like (very pseudocode here) `ttblit.serial.user("message")`

Daft-Freak, 2025-01-27:
There's raw serial access since e0133c9be67945752380e95959a8e97c953478bf, but kinda meant for "internal" use. (Is used in the slightly hacky MicroPython port)

Might still want a way to receive something without taking over the entire interface though, so not closing...