32blit/32blit-sdk #596: Update docs to refer to the proper Debian version

IkerGarcia, 2021-02-13:
Hi, 

I was following the guide but didn't work properly under Debian WSL stretch. Asked in discord, upgrade to buster and now works ok. It should be pointed out in the docs.

Daft-Freak, 2021-02-13:
Quoting myself:
> Hmm, should probably document required versions of stuff. Linux/WSL docs pretty much assume "recent-ish Debian-based"

Some required versions:
- gcc-arm-none-eabi 7.x
- CMake 3.9
- Python 3.6

So... Ubuntu 20.04 (or even 18.04 with a newer toolchain), but Debian buster (not even stretch + backports)

Daft-Freak, 2021-02-21:
Adding this here so I don't forget: gcc-arm-none-eabi 10.x doesn't work... need to debug that.

Daft-Freak, 2021-05-28:
Made some improvements in #660

ahnlak, 2023-01-31:
While this is pretty ancient history, I'm guessing that since https://github.com/32blit/32blit-sdk/commit/727e0570b38eb86d90f6dc4a938f60499f4af88a we should be able to close this?

Daft-Freak, 2023-01-31:
Hmm, might need to change the debian requirement to bullseye. (Updated the GCC requirement in 96b4618ac2b2222beeaf57005928199361f01acc... and buster does not have 8.x, oops.)

This is entirely because I don't have a buster install to check though.

PR #887 (2025-01-27): docs: update required Debian version
buster only has gcc-arm-none-eabi 7. Also bump the max as I'm currently using 14.

Technically already fixed, then broken again but... Fixes: #596
closed 2025-01-27 by Daft-Freak