32blit/32blit-sdk #597: Multiplayer can only receive one message per render()

ali1234, 2021-02-14:
This code sends one message per update().

```
#include <cstring>

#include "multiplayer.hpp"

#include "engine/multiplayer.hpp"

using namespace blit;

uint32_t sent = 0;
uint32_t received = 0;
uint32_t frames = 0;

void on_message(const uint8_t *data, uint16_t len) {
    received++;
}

void init() {
    message_received = on_message;
    enable_multiplayer();
}

void render(uint32_t time_ms) {
    char msg[100];
    sprintf(msg, "sent: %d\nreceived %d\nframes: %d", sent, received, frames);
    screen.pen = Pen(0, 0, 0);
    screen.clear();
    screen.alpha = 255;
    screen.pen = Pen(255, 255, 255);
    screen.text(msg, minimal_font, Point(0, 30));
    frames++;
}

void update(uint32_t time_ms) {
    auto out = "Hi!";
    send_message((uint8_t *)out, strlen(out));
    sent++;
}

```

Result:

[image]

Daft-Freak, 2021-02-14:
Hmm, only affects SDL due to it only doing updates every 20ms (https://github.com/32blit/32blit-sdk/blob/master/32blit-sdl/System.cpp#L199)... and Multiplayer::update only handling one message before it returns. 

Throwing a loop in there helps. https://github.com/Daft-Freak/32blit-beta/commit/3801a950b9f8720061f8dc9f247d5ffb8dd96b69

ali1234, 2021-02-27:
Fixed in #602

closed 2021-02-27 by ali1234