32blit/32blit-sdk #631: Game metadata corrupt in launcher after flashing a blit

ali1234, 2021-03-08:
To reproduce:

1. Be in the launcher.
2. `32blit install` any blit with metadata.
3. Select the thing you just flashed in the launcher.

Result: The metadata for the newly flashed blit is corrupted in the launcher. The image can be blank or corrupted, and the rest of the metadata may be missing, including the game name which might be replaced by something like "2".

Attempting to launch the blit from the launcher at this stage may lead to the game and/or launcher crashing.

Restarting the launcher by selecting "exit game" from the menu will clear the problem, and the blit will be displayed correctly and loading will work.

Gadgetoid, 2021-03-09:
This was briefly mentioned in https://github.com/32blit/32blit-sdk/pull/616

Basically the launcher no longer resets after a file is uploaded, so it never refreshes the file list. The previous location of the game is now empty (if it has been replaced/erased) so it's probably reading erased flash sectors as metadata and this tends to result in unpredictable behaviour.

The firmware needs some way to signal to the launcher that the flash contents have changed. Previously the re-launch swept this problem under the rug.

Daft-Freak, 2021-03-09:
Almost forgot about that one. Quick fix would be "resetting" the launcher by re-calling it's init func:
```c++
if(blit_is_launcher_running()) {
  auto game_header = (BlitGameHeader *) (qspi_flash_address + persist.last_game_offset);
  auto init = (BlitInitFunction)((uint8_t *)game_header->init + persist.last_game_offset);
  init(persist.last_game_offset);
}
```
(Possibly tidier than that, `init` pointer currently isn't stored anywhere)

PR #632 (2021-03-11): Patch launcher reset after flash
Tries to fix #631 but has its own drawbacks.

Still crashes after about 45 game flashes, but ensures the launcher list is re-loaded immediately after the flash and avoids listing a game that has been deleted.
closed 2021-03-11 by Gadgetoid