32blit/32blit-sdk #681: dfu-util: Invalid DFU suffix signature

ali1234, 2021-06-06:
When flashing the latest firmware on linux I get this warning message:

```
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
```

I am not sure what this means but we should look into it to avoid it breaking at some point in the future.

I suspect it means there is some invalid header on the firmware.dfu.

Gadgetoid, 2021-06-14:
The suffix is this portion here: https://github.com/32blit/32blit-sdk/blob/dd9d68df7246eb220abc71168d7e5ceee4cf3de6/tools/dfu#L117-L122

Which, according to this datasheet: http://rc.fdr.hu/UM0391.pdf

● The bcdDevice field gives the firmware version contained in the file, or 0xFFFF if ignored.
● The idProduct and idVendor field give the Product ID and Vendor ID respectively of the device that the file is intended for, or 0xFFFF if the field is ignored.
● The bcdDFU field, fixed to 0x011A, gives the DFU specification number. This value differs from that specified in standard DFU rev1.1.
● The ucSignature field contains a fixed string of three unsigned characters (44h, 46h, 55h). In the file they appear in reverse order, allowing valid DFU files to be recognized.
● The bLength field, currently fixed to 16, gives the length of the DFU Suffix itself in bytes.
● The dwCRC (Cyclic Redundancy Check) field is the CRC calculated over the whole file except for the dwCRC data itself

And to meet the conditions for the CRC:

https://github.com/32blit/32blit-sdk/blob/dd9d68df7246eb220abc71168d7e5ceee4cf3de6/tools/dfu#L125-L130

This all *looks* right, insofar as there isn't much to get wrong.

Maybe - contrary to the comments at the top of the DFU tool - the suffix should be included in the file length: https://github.com/32blit/32blit-sdk/blob/dd9d68df7246eb220abc71168d7e5ceee4cf3de6/tools/dfu#L55

Gadgetoid, 2021-06-14:
P.S. What's your DFU Util version and flash command?

On Ubuntu I see a different flavour of wrong:

```
dfu-util -D firmware.dfu -d 0x483:0xdf11 -a 0
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
dfu-util: Error: File ID 0483:0000 does not match device (0483:df11 or 0483:df11)
```

IE: The DFU file has 0000 as the Product ID so it will refuse to flash.

ali1234, 2021-06-14:
Okay I found the problem. I typed `firmware.bin` instead of `firmware.dfu` and apparently this actually works, but produces the warning because no suffix.

ali1234, 2021-06-14:
0000 is probably because of this:

https://github.com/32blit/32blit-sdk/blob/master/tools/dfu#L190

ali1234, 2021-07-21:
Over in the new DFU tool the default is still 0 for product ID:

https://github.com/32blit/32blit-tools/blob/master/src/ttblit/core/dfu.py#L121

Gadgetoid, 2021-07-22:
That's definitely an oversight, should be `df11`. Me abandoning DFuSe in favour of Linux has not helped this.

Daft-Freak, 2025-01-27:
I think the remaining issues here are that the docs tell you to use the .bin, which gives the signature warning and that using the .dfu is differently broken (but easily fixed: https://github.com/32blit/32blit-tools/pull/106)