32blit/32blit-sdk #734: Another World does not launch

davidskeck, 2021-11-27:
I cannot get Another World to launch on either my 32Blit or desktop.

[image]

mslinklater, 2021-11-28:
Same here. It's failing to load 'memlist.bin' and then bailing due to missing resources.

Daft-Freak, 2021-11-28:
You need to get a copy of the original game from somewhere. Though it doesn't get very far if you do have the data either as the heap is too small (https://github.com/32blit/32blit-sdk/blob/master/examples/another-world/virtual-machine.hpp#L15). Increasing that prevents it crashing. but still a bit broken.

Gadgetoid, 2022-01-07:
This has now been moved to it own repository, with some more instructions about what data files are required - https://github.com/32blit/32blit-another-world

It is, indeed, very broken.