32blit/32blit-sdk #736: picosystem saves are corrupted

protogenraymond, 2021-11-30:
i was working on a modded build of Super Square Bros, and when i ran a function to reset my data, it crashed. after i restarted the picosystem, saving did nothing.

when i plugged it into my computer, i deleted the .blit folder, but now it just only has 1 file inside the Super Square Bros folder- save0. there's nothing inside this file, so im pretty sure i messed up my picosystem or there's a bug with some kind of metadata.

i tried loading another build of Super Square Bros, but it still has the same problems. i will be attempting to change the metadata file to see if that does anything but i don't think that'll change much.

protogenraymond, 2021-11-30:
update: modifying the metadata didn't help at all. it puts it into it's own new folder but there still is no data being saved.

FKHals, 2022-10-13:
I just experienced a similiar problem on the picosystem where saving did not work most probably due to file system corruption (caused by my own code).

I fixed it by repairing the file system using standard linux utilities:
```shell
# Identify the drive (search for the one of size 4.00 MiB), for example `/dev/sda`.
fdisk -l
DRIVE=/dev/sda

# Unmount the corresponding drive
umount $DRIVE

# Use fsck to check and repair the drive:
sudo fsck -p $DRIVE
```
Now everything should work again and to regain storage space you can simply remove the fsck revocery files (FSCK*.REC) that have been created on the drive, assuming that no data recovery is necessary.