32blit/32blit-sdk #776: `make firmware` fails with Arch Linux default arm-none-eabi-gcc + arm-none-eabi-newlib

pogojotz, 2022-07-23:
More precisely the linkage fails:
```
[...]
/usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/firmware.dir/firmware.cpp.obj: in function `std::_Vector_base<unsigned long, std::allocator<unsigned long> >::_M_allocate(unsigned int) [clone .isra.0]':
firmware.cpp:(.text._ZNSt12_Vector_baseImSaImEE11_M_allocateEj.isra.0+0x10): undefined reference to `std::__throw_bad_array_new_length()'
/usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: /tmp/ccWLFRW6.ltrans2.ltrans.o: in function `std::_Vector_base<void*, std::allocator<void*> >::_M_allocate(unsigned int) [clone .isra.0]':
<artificial>:(.text._ZNSt12_Vector_baseIPvSaIS0_EE11_M_allocateEj.isra.0+0x10): undefined reference to `std::__throw_bad_array_new_length()'
collect2: error: ld returned 1 exit status
make[3]: *** [firmware/CMakeFiles/firmware.dir/build.make:317: firmware/firmware.elf] Error 1
make[2]: *** [CMakeFiles/Makefile2:2788: firmware/CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2795: firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [Makefile:1138: firmware] Error 2
```

A quick search suggests this might be connected to GCC v11+, e.g. [[1]](https://github.com/JuliaLang/julia/issues/40703)

My installed packages:
```
$ pacman -Qs arm-none-eabi 
local/arm-none-eabi-binutils 2.38-1
    A set of programs to assemble and manipulate binary and object files for the ARM EABI (bare-metal) target
local/arm-none-eabi-gcc 12.1.0-1
    The GNU Compiler Collection - cross compiler for ARM EABI (bare-metal) target
local/arm-none-eabi-gdb 12.1-1
    The GNU Debugger for the ARM EABI (bare-metal) target
local/arm-none-eabi-newlib 4.2.0.20211231-1
    A C standard library implementation intended for use on embedded systems (ARM bare metal)
```

pogojotz, 2022-07-23:
I just tested the official precompiled toolchains [gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2](https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2) and [gcc-arm-11.2-2022.02-aarch64-arm-none-linux-gnueabihf.tar.xz](https://developer.arm.com/-/media/Files/downloads/gnu/11.2-2022.02/binrel/gcc-arm-11.2-2022.02-aarch64-arm-none-linux-gnueabihf.tar.xz)

I had the same issue with version 11.2, but not with 10.3. Anyone experienced the same?

Daft-Freak, 2022-07-24:
Yeah, I've run into this... It's probably our [custom prebuilt stdlibs](https://github.com/32blit/stdlibs) being a bit too outdated for GCC 11+.

pogojotz, 2022-07-26:
Ow that make sense. Shouldn't that documented somewhere? Or better yet, the stdlibs updated? I spent like 2 hours doubting myself 🤣

Daft-Freak, 2023-01-26:
stdlibs updated in #788, appears to build fine with GCC 8-12

closed 2023-01-26 by Daft-Freak