32blit/32blit-sdk #848: Can’t build for 32Blit on all Platforms

lummi01, 2024-05-27:
The new created .blit files are corrupted for the 32Blit Console.
Tested on Win10, Win11, Linux Mint and Raspberry OS.
Local builds are ok.

Daft-Freak, 2024-05-27:
https://github.com/32blit/32blit-tools/pull/103 should help. `pip install bitstring==4.0.0` may also be a workaround (assuming bitstring was installed through pip with the tools).

I haven't (yet) found anything other than images to be broken. I've only checked a few examples and picovision-3d though.

lummi01, 2024-05-27:
'bitstring==4.0.0' fix the problem with the splash screen but the .blit file is still corrupted.
Even the pure 32Blit-Boilerplate without any changes produce this failure.
[image]

Daft-Freak, 2024-05-27:
Hmm, I think the most likely cause of that error message is an incompatible firmware version... and I may have bumped the API version since the last release. So if you're using the latest SDK git you'll need to build/install the firmware.

(Not surprising that I forgot that my device wasn't running released firmware...)

lummi01, 2024-05-27:
That was exactly the problem. Sorry for my ignorance...

Daft-Freak, 2024-05-27:
Well, it's not the most helpful error message.

I should really finish https://github.com/32blit/32blit-sdk/pull/822 so it can be more specific. :thinking: 

(... and maybe do something crazy like make a release that has PicoVision support)

Gadgetoid, 2024-05-29:
> (... and maybe do something crazy like make a release that has PicoVision support)

:eyes:

Daft-Freak, 2024-06-22:
Splashes should be fixed with latest release of the tools and the launcher should be more clear about version incompatibility in the future.

Plus there isn't an unreleased API bump any more.

closed 2024-09-20 by Daft-Freak