32blit/32blit-sdk #853: Can't build for picosystem

Quentie, 2024-07-04:
Hi I'm trying to follow https://github.com/32blit/32blit-sdk/blob/master/docs/pico.md to setup a Dockerfile to build for my new picosystem. But can't get it working :( Could anyone give me a tip?

**Here my Dockerfile:**

`FROM ubuntu:24.10

RUN apt update \
    && apt install --yes --no-install-recommends \
       git gcc g++ gcc-arm-none-eabi cmake make \
       python3 python3-pip python3-setuptools \
       libsdl2-dev libsdl2-image-dev libsdl2-net-dev unzip \
    && pip3 install 32blit --break-system-packages \
    && git clone https://github.com/32blit/32blit-sdk \
    && cd 32blit-sdk \
    && mkdir build \
    && cd build \
    && cmake .. -D32BLIT_DIR=`pwd`/.. -DPICO_SDK_FETCH_FROM_GIT=true -DPICO_EXTRAS_FETCH_FROM_GIT=true -DPICO_BOARD=pimoroni_picosystem \
    && make \
`
If I build the docker image using `docker build - < Dockerfile` I get 

`
-- Build files have been written to: /32blit-sdk/build
1005.8 [  0%] Building CXX object 32blit/CMakeFiles/BlitEngine.dir/audio/audio.cpp.obj
1005.9 In file included from /32blit-sdk/32blit/audio/audio.cpp:4:
1005.9 /32blit-sdk/32blit/audio/../engine/engine.hpp:3:10: fatal error: cstdint: No such file or directory
1005.9     3 | #include <cstdint>
1005.9       |          ^~~~~~~~~
1005.9 compilation terminated.
1005.9 make[2]: *** [32blit/CMakeFiles/BlitEngine.dir/build.make:76: 32blit/CMakeFiles/BlitEngine.dir/audio/audio.cpp.obj] Error 1
1005.9 make[1]: *** [CMakeFiles/Makefile2:2223: 32blit/CMakeFiles/BlitEngine.dir/all] Error 2
1005.9 make: *** [Makefile:156: all] Error 2
`

It works like a charm if I build for linux and not the picosystem:
`
FROM ubuntu:24.10

RUN apt update \
    && apt install --yes --no-install-recommends \
       git gcc g++ gcc-arm-none-eabi cmake make \
       python3 python3-pip python3-setuptools \
       libsdl2-dev libsdl2-image-dev libsdl2-net-dev unzip \
    && pip3 install 32blit --break-system-packages \
    && git clone https://github.com/32blit/32blit-sdk \
    && cd 32blit-sdk \
    && mkdir build \
    && cd build \
    && cmake .. -D32BLIT_DIR=`pwd`/.. -DPICO_SDK_FETCH_FROM_GIT=true -DPICO_EXTRAS_FETCH_FROM_GIT=true -DPICO_BOARD=pimoroni_picosystem \
    && make \
`

Daft-Freak, 2024-07-04:
Hmm, I'm guessing that with `--no-install-recommends` you're going to need to specify the stdlib packages. (`libnewlib-arm-none-eabi`/`libstdc++-arm-none-eabi-newlib`?)

Quentie, 2024-07-04:
Thanks @Daft-Freak! That worked like a charm!

closed 2024-07-04 by Quentie