32blit/32blit-sdk #860: Remove decode_jpeg_* APIs?

Daft-Freak, 2024-09-20:
I've been thinking about this a bit:
- As far as I can tell, `decode_jpeg_buffer` has two users (the example and my video player demo), `decode_jpeg_file` has none
- They're a big chunk of code in the STM32 firmware (~6k text, ~5k bss)
- Not even implemented in `32blit-pico`
- sdl2-image keeps causing dependency issues
- It would probably be more useful to have something that wraps the same decoder everywhere on the user side of the API :thinking:

Gadgetoid, 2024-09-20:
I wonder if anyone else has used this? JPEG is definitely one of the worst ways to get image data onto a microcontroller and we have an explicit pipeline for converting assets!

ali1234, 2024-09-21:
I would much rather see that valuable firmware space used for something that is widely useful and benefits from being fast, like more drawing routines (#32), higher level multiplayer code, or more flexible menu system, or maybe even something like stagelib. Even just another font or some standard button sprites would be more useful.