32blit/32blit-sdk #880: ccache: Cache is not building/restoring/saving

Gadgetoid, 2024-12-03:
I've been debugging loooooong CMake build times on pimoroni-pico with ccache being the culprit. I remembered I probably inherited the ccache code from here, so when I figure out how to fix it, should probably fix it here too!

Check any action with ccache and you'll see a cache miss, and a fail to upload the resulting cache, it looks like ccache has changed to not use `/home/runner/.ccache` (`~/.cache`) by default anymore.

Gadgetoid, 2024-12-03:
`On non-Windows systems, the default is $HOME/.ccache if such a directory exists`

`if such a directory exists`

🤔  we have a good ol' fashioned paradox.

Daft-Freak, 2024-12-03:
Maybe we should set CCACHE_DIR explicitly like the mac build does... though it seems that one isn't working either.

Gadgetoid, 2024-12-03:
Well that worked (and is extremely cursed!)

https://github.com/pimoroni/pimoroni-pico/pull/1032/commits/92902439107135407f911445d2441bbd27269785

Gadgetoid, 2024-12-03:
My best sleuthing suggests that this change happened with ccache 4.0 on 2020-10-18.

I think we *might* have been riding on having an existing cache - which implicitly created the directory when restored - for much longer. It wouldn't break until cache misses started landing, and all branches get access to the default branch cache so that could have taken some time (long enough between main builds for the cache to expire?). That's my best theory for "why stable not stable" anyway.

Daft-Freak, 2024-12-03:
This also seems to work: https://github.com/32blit/32blit-sdk/commit/9d6adb338d462e061c09781ed50bf636ff899c87

Relying on a fallback was probably always a bad idea, it's just weird that it suddenly broke now...

PR #886 (2025-01-27): Fix ccache path
Probably should've actually merged this fix.

Fixes: #880