32blit/32blit-sdk #890: macOS: EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))

Gadgetoid, 2025-01-31:
Because I will probably forget how to fix this as fast as I can blink...

(I don't know how to fix it yet)

Daft-Freak, 2025-01-31:
Need to (re-)sign the app with something like `codesign --force --deep -s [cert id] blit-kart-0.1.1-Darwin/bin/Super\ Blit\ Kart.app`

That is, after doing all the xcode stuff to get a working certificate... (Which I can't remember, just that it didn't work the way it was supposed to)

Gadgetoid, 2025-02-01:
That sounds vaguely familiar! I think I jumped through those hoops before and then promptly erased it from memory.

Gadgetoid, 2025-02-01:
Hmmm as an aside opening `Launcher.app` and pressing right immediately segfaults 😆

Daft-Freak, 2025-02-02:
> Edit2: Okay it's the _installed_ binaries that fail, the originals in the build tree work fine.

Think it's the dependency fixups invalidating any existing signatures. (I'm not sure if the executable itself even has one, or if it does where it's getting it from...)

> Hmmm as an aside opening `Launcher.app` and pressing right immediately segfaults 😆

Probably SDL having `nullptr` for all the launcher-related API. (I feel like I have patches for that...somewhere?)

Gadgetoid, 2025-02-02:
> Probably SDL having `nullptr` for all the launcher-related API. (I feel like I have patches for that...somewhere?)

I’m sure I remember something along these lines too! 🤣

Daft-Freak, 2025-02-05:
Hah, https://github.com/Daft-Freak/32blit-beta/tree/sdl-launch

(This has very little to do with the original issue, but I had to go find it...)